TeXitoi/keyseebee #1: Plate files in DFX format

octol, 2020-10-18:
I'd be quite interested in trying metal switch plates on my KeyBeeSee. Something that still exposed the components in the middle would be preferable. If we had DXF files available this would make it easy for users to have plates cut

TeXitoi, 2020-10-20:
It would also be great to have SVG to be presented in https://github.com/jhelvy/splitKbCompare

I'll try to do a scad when I have a bit of time, and 3D print it to be sure that's the good size.

eropleco, 2020-10-30:
Having plates would be quite nice indeed, and I think making the plates as another set of PCBs might work too, since it'd be a very economical solution compared to getting a metal plate cut.

TeXitoi, 2020-10-30:
Well, the main concept is minimalism, only a PCB. There is nothing in the PCB to support any plate.

octol, 2020-10-30:
Correct me if I'm wrong by I think what was meant was to have FR4 plates, which would be a lot cheaper to produce. Not to use a PCB as plate?

octol, 2020-10-30:
I too would actually prefer to have an FR4 plate, too keep the black theme I've got going on mine currently

TeXitoi, 2020-10-30:
Yeah, I mean the PCB doesn't help in using a okate, you'll need a floating PCB, and a wide plate to connect with a bottom plate. You can't have any screw on the PCB.

octol, 2020-10-30:
I was just thinking of a floating plate, only attached to the switches. This is what I have now on my ErgoTravel and it works great. Gives extra rigidity without making the board any taller

octol, 2020-10-30:
So the plate would have the exact same outer dimensions as the pcb, except for the inner area where all the components are, which I think should be left open as I like the exposed components esthetics

TeXitoi, 2020-10-30:
That's who I would do the design, using openscad.

TeXitoi, 2020-11-03:
https://github.com/TeXitoi/keyseebee/tree/master/cad

[image]

Note that the left and the right plate are not exactly the same, there is a small offset of the right cluster. To be corrected in v0.3

TeXitoi, 2020-11-03:
I'll close this when keyseebee is on https://github.com/jhelvy/splitKbCompare

TeXitoi, 2020-11-03:
https://github.com/jhelvy/splitKbCompare/issues/28

TeXitoi, 2020-11-03:
You can export in DXF using openscad. You can also change the switch hole size to match the tolerance you need.

For ordering FR4 plates, I suppose you need to construct a gerber from the file, using the svg as the cut layer. You may add some copper zone, and can do some crazy art using the different layers. https://wiki.ai03.com/books/pcb-design/page/pcb-guide-part-6---fill-zones-decoration-and-production You may have extra fees because of complicated cuts.

eropleco, 2020-11-06:
> Yeah, I mean the PCB doesn't help in using a plate, you'll need a floating PCB, and a wide plate to connect with a bottom plate. You can't have any screw on the PCB.

Thank you for making the plate files! And yes, this has always been my intention, to just have a floating plate to keep the switches in place, mostly for easy cap removal.

TeXitoi, 2020-11-06:
@eropleco How does a plate help for cap removal? That's not a hotswap board.

And, out of curiosity, do you have a keyseebee? Maybe you sourced the PCB yourself?

octol, 2020-11-06:
Removing the MBK keycaps on mine you have to be very very careful to not rip the solder joints as the caps sits very firmly on the switches. A plate might help here?

Sort of related, I noticed that on my Ferris I don't have that problem since the switch alignment holes are slightly slightly smaller making the switches sit firmly anchored on the pcb, even without them being soldered. Not sure if it's because of production tolerances the holes were designed to be smaller?

TeXitoi, 2020-11-06:
Did you removed a switch while trying to remove a keycap? If yes, where the join break? Maybe you don't put enough solder?

The ferris is a choc only board. KeySeeBee is also MX compatible, and the plastic legs are not exactly the same, thus there is 2 holes for each lateral plastic legs that are a bit offset. You then have a less strong press fit.

octol, 2020-11-06:
By pulling carefully while wiggling the keycap puller I managed to remove the keycaps without removing any switches

Since the solder joints are only on one side, pulling the keycaps straight up means the switch will break away from the pcb on the other non-soldered side, a bit like opening a hinge. So the amount of solder has no real effect there

Didn't know the alignment holes for MX and Choc were slightly different, and yeah looking closer I see what you mean

TeXitoi, 2020-11-06:
OK, I see what you mean. The transparent keycaps are not so hard to remove, so I have no problem removing them. Maybe a plate can help then.

eropleco, 2020-11-06:
> And, out of curiosity, do you have a keyseebee? Maybe you sourced the PCB yourself?

I just built one recently but haven't yet finished soldering the switches since the plate is now a thing. I'll post some images eventually when I'm done with it. 

> How does a plate help for cap removal? That's not a hotswap board.

The key caps on the choc switches are very difficult to remove for me without exerting a huge amount of force on the legs, and a plate will help with this problem by holding the switches in place. I will need to make some changes to the design so the choc switches actually clicks into the plate though.

[image]

TeXitoi, 2020-11-06:
@eropleco So you sourced yourself a v0.2? Can I validate that everything is functional on the gerber page?

The place should match the choc keycaps, what you mean by "make some changes to the design so the choc switches actually clicks into the plate"? You can change easily the different values, as switch holes and (for the stl) plate thinkness by editing the parameters at the top of the plate.scad file.

eropleco, 2020-11-06:
Yes, I used v0.2 when ordering the PCB. I was about to post an issue here to tell you about it but since I haven't actually finished the board, I thought I'd hold that off for a bit before I can verify everything. So far what I *can* confirm is that all of the footprints are correct, I can flash the firmware, and the keyboard does send keycodes to the computer. I didn't test *all* of the keys, nor have I tested UART communication yet due to the lack of a usable TRRS audio cable.

About the modifications, I need to change the thickness of the plate, and the size of the opening. According to the illustration in the middle, the thickness is 1.3mm, so the standard 1.2mm PCB thickness should work quite well here. The opening should be 13.8mm. 

I'm not familiar with openscad, I see I can change both of these parameters in the scad source file, but how do I get the left side of the plate?

TeXitoi, 2020-11-06:
I added comments in the scad file, tell me if that's enough to tune for a non openscad user.

eropleco, 2020-11-06:
Thank you, the comments are very helpful. I'll share some pictures when I get my plates manufactured.

TeXitoi, 2020-11-06:
For the holes, by the datasheet, they must be between 13.8 and 14.5. Generally, you add a bit of tolerance, so 14mm is a standard choice.

eropleco, 2020-11-06:
I measured the switch with a digital caliper and got 13.8mm too, so that's what I'm inclined to use. The actual body of the switch is 13.6mm wide, the 13.8mm measurement includes these several tiny lips. 14mm is dangerously close to the width of the 4 clips, which measures just 14.3mm here.
[image]

TeXitoi, 2020-11-06:
OK, I'd be interested in feedbacks on the plate one time you've done and tested it @eropleco.

eropleco, 2020-11-19:
@TeXitoi This is is the end result. My soldering job weren't perfect but I tried (notice the slightly crooked diodes). The plate is 1.2mm anodized aluminium, and I used 13.9mm for the hole diameter. The manufacturing precision wasn't great so a couple of holes were a bit loose, and there was one that is a bit too tight, so I had to shave the switch down a tiny bit. Overall it's not too bad though, I'm pretty happy about how it turned out in the end.

[image]
[image]

TeXitoi, 2020-11-19:
Ho, really great! Out of curiosity, how much the plates did cost?

TeXitoi, 2020-11-19:
Can I use the photos in the readme?

eropleco, 2020-11-19:
The plates cost me about $20 in total at a local shop, and yes, you can use the images in the readme :)

TeXitoi, 2020-11-19:
I've updated the README.md and the pcb/gerbers/CHANGELOG.md accordingly. Thanks.

$20 is not so expensive, would think anodized aluminium to be more expensive. It looks great!

And your soldering job is really great! Much better than mine!

TeXitoi, 2020-11-19:
@eropleco I'm sure https://www.reddit.com/r/ErgoMechKeyboards/ would appreciate a photo of your build!

TeXitoi, 2020-11-19:
@eropleco also, what did you do for homing? I can see a dot in the keycap.

eropleco, 2020-11-19:
Thanks for the compliment, I was being extra slow and careful while making this so it'd look neat. Washed the board a couple of times before putting on the switches, too. 

Regarding that dot, since I wasn't able to source any existing key caps with a bump on it, I just poked a hole on the cap with a heated straight pin, clipped the head off with about 0.5mm of the stem left, and gently pushed that bit in while heating it with a soldering iron. This way I'd be left with a round metal bump that is semi-embedded within the plastic. A friend suggested epoxy, but my two component epoxy has expired, and tbh I'm not sure how would I be able to shape a tall, uniform bump with it, so I didn't try that idea. A small SMT part that doesn't have solder exposed on top would probably work too, such as a crystal.
 
[image]

TeXitoi, 2020-11-20:
keyseebee is now live on https://jhelvy.shinyapps.io/splitkbcompare/

closed 2020-11-20 by TeXitoi