TeXitoi/keyseebee #3: Clarify documentation about flashing

octol, 2020-10-29:
Building the firmware and flashing the board was pretty straightforward, however something that would be nice to clarify in the documentation is if it's necessary to flash both sides (is it? I've not done it and I have not noticed any issues?).
Also is my assumption correct in that you can use either side as primary?

(Sometimes I think issues/questions like these better belong in a forum, alternatively Discord. For example I'd be happy to raise a PR if I knew the answer to the above.)

TeXitoi, 2020-10-29:
if you only change the layout, you can only flash the side you plug. You can have 2 different layout on each side, and plugging in one side will use the layout on this side.

Basically, each side send coords of press and release of each key (after debouncing) via the TRRS cable and also to the layout manager. So the layout manager receive key event via another task on the same microcontroller and via UART on the TRRS cable.

Then, the layout send to USB the events if USB is connected.

TeXitoi, 2020-10-29:
If you want a chat, I'm present on a few embedded rust chan on matrix.org. I will not go to discord because that's closed, and I don't want to multiply networks I connect. (Also, there is a matrix client on my ungoogled smartphone, and that's not the case for the other protocols)

TeXitoi, 2020-10-29:
In theory, you can plug each side to a given computer, and you'll type on the 2 computers at the same time. But I affraid to try it as it would connect the usb power of the 2 computers in parallel. In theory, this is OK, but I don't think it's a good idea.

octol, 2020-10-29:
Hey that's pretty cool! Sounds great for being to toggle back and forth between different layouts, for example now that I'm testing home row mods but still want to easily be able to switch back

eropleco, 2020-10-30:
> In theory, you can plug each side to a given computer, and you'll type on the 2 computers at the same time. But I affraid to try it as it would connect the usb power of the 2 computers in parallel. In theory, this is OK, but I don't think it's a good idea.

Regarding this idea, I think you can simply add a switch on one side to disconnect VCC, and then add an ADuM3201 for isolating the UART communication: https://www.analog.com/media/en/technical-documentation/data-sheets/ADUM3200_3201.pdf

PR #4 (2020-11-04): Add note about which side needs flashing
Resolves #3
closed 2020-11-04 by TeXitoi