TeXitoi/keyseebee #6: 0.1 LDO fix

keyboard-magpie, 2020-12-01:
A friend and I have some 0.1 keyseebee PCBs we're building up- is it possible to see a hi res photo, or have a description of the fix needed? I can see in some photos the LDO is at an unusual angle.

TeXitoi, 2020-12-01:
The alone leg must be at 1 o'clock. More info here: https://github.com/TeXitoi/keyseebee/blob/master/pcb/gerbers/CHANGELOG.md#v01-2020-08-02

Why did you printed the v0.1 and not v0.2?

keyboard-magpie, 2020-12-01:
Got it! :D Multimeter and such.
No idea why  did, i think he downloaded just before 0.2 was released.
Up and running now though.

https://cdn.discordapp.com/attachments/760085292048187442/783447743632244796/image0.jpg

closed 2020-12-01 by keyboard-magpie

TeXitoi, 2020-12-01:
Exactly. I hope you've soldered the diodes in the good direction. If not, that should be fixable in the firmware.

keyboard-magpie, 2020-12-01:
Oh, hadn't spotted that one! Presume they're opposite to silkscreen marking then?

TeXitoi, 2020-12-01:
yup, dash up.

keyboard-magpie, 2020-12-01:
Perfect thanks. Will pass this on.

TeXitoi, 2020-12-01:
yup, dash up.