TeXitoi/keyseebee #8: USB enumeration

keyboard-magpie, 2020-12-04:
Me again, so sorry.

Both halves of my 0.1 build now present bootloader.
However when flashed with keyberon from this repo neither enumerate/appear in lsusb. 
Running dmesg on linux shows enumeration error 71.

I tried flashing a known working qmk f072 bin, which does enumerate fine and presents a usb keyboard.

Any thoughts? Having looked through the commits to keyberon I couldn't see anything that changed clock speed tree etc that might affect this.

Many thanks

TeXitoi, 2020-12-04:
https://github.com/TeXitoi/keyberon/pull/36

So 2 solutions:
 * switch to rust 1.47
 * update to keyberon master

The first solution is the simplest.

keyboard-magpie, 2020-12-04:
Swapped to rust 1.47 and all good!

gwenzek, 2021-01-04:
I also had the same issue
upgrading to https://github.com/TeXitoi/keyseebee/tree/keyberon-master (8654780
)
works well with `rustc 1.49.0 (e1884a8e3 2020-12-29)`

camrbuss, 2021-02-06:
I too suffered from the generic_array panic on the master branch with rustc 1.48.0

Just checking out the `keyberon-master` branch as gwenzek said solved the panic by grabbing keyberon master branch. Can something get merged into the keyseebee master branch so that new comers (like me) don't build a panicking firmware?

TeXitoi, 2021-02-06:
Yeah, i have to release keyberon v0.2

TeXitoi, 2021-09-15:
Keyseebee firmware now use the master keyberon.

closed 2021-09-15 by TeXitoi