TeXitoi/keyseebee #10: Help with keyseebee keyberon firmware

MangoIV, 2021-06-14:
Hi! I've had a hard time understanding your code for the communication between the halves. Would you give me an insight on how it's happening? I'm especially curious about this function: 
```rust
    #[task(binds = USART1, priority = 5, spawn = [handle_event], resources = [rx])]
    fn rx(c: rx::Context) {
        static mut BUF: [u8; 4] = [0; 4];

        if let Ok(b) = c.resources.rx.read() {
            BUF.rotate_left(1);
            BUF[3] = b;

            if BUF[3] == b'\n' {
                if let Ok(event) = de(&BUF[..]) {
                    c.spawn.handle_event(Some(event)).unwrap();
                }
            }
        }
    }
```

What exactly is written to and read from the Buffer? 
Thanks in advance :)

TeXitoi, 2021-06-14:
The function is called (by the harware via an interruption) each time a character is ready to be read.

We get this character in the if let.

We push the character at the end of the buffer, removing the first character to make place.

If the last character is \n then our command is complete. We parse it and send the deserialized event to the layout handler.

TeXitoi, 2021-06-14:
For the serialization of an event see the ser function. For the deserialization see the de function.

MangoIV, 2021-06-14:
Thanks for your quick answer and sorry if the question is stupid. Now that I know what the functions are doing it's much easier on the eyes.. :D

closed 2021-09-15 by TeXitoi