TeXitoi/keyseebee #11: The Computer Language Benchmarks Game (CLBG)

evertondanilo, 2021-06-17:
Hi, TeXitoi. Hope you are keeping well.
Did you contribute to port binary-trees algorithm to Rust in the CLBG project? I want to compile it to Wasm.

TeXitoi, 2021-06-17:
A long time ago, yes https://github.com/TeXitoi/benchmarksgame-rs/blob/master/src/binary_trees.rs

evertondanilo, 2021-06-17:
Hi @TeXitoi thanks for your quick reply. I'm working on a research project for my masters. The idea is to compare Wasm with JS in the browser. How fast it would be? Do you think CLBG algs can be compiled to Wasm?

TeXitoi, 2021-06-17:
you can solve the benchmarkgames in wasm, but you may not be able to take directly the version on the site as they depend on features not availlable on wasm (simd, std...).

I suspect it will be faster than the js version, but a well written js and rust implementation might be of similar speed for some benchmarks.

closed 2021-09-15 by TeXitoi