TeXitoi/keyseebee #14: IDEA: Collective order of presoldered kits

555wolfman, 2021-09-27:
This keyboard looks so slick! Well done!

I am interested in getting some kits, but i am afraid smd soldering is way beyond my skills.
As i know PCBway, JLCPCB and Elecrow offer assembly  services when ordering a PCB. 

So my question is whether someone plans to order presoldered PCBs. I would like to join in.
And i am sure i am not the only one.

thanks in advance

Olaf

TeXitoi, 2021-09-27:
The files are not ready to send to a SMT assembly service, it lacks a proper BOM I suppose. Open to contribution about that.

Also, note that MCU are out of stock everywhere, so it will be difficult to find STM32F072 on the SMT assembly service, and ordering SMT without the MCU is quite useless, as it is one of the most difficult component to solver (with the USB connector).

TeXitoi, 2021-09-27:
If it becomes a thing, I can proxy for europe.

555wolfman, 2021-09-27:
thanks for offering that proxy service in case of a solution.

I have heard about that supply problem.

stm32f072c8t and f072cbt are available at aliexpress as well as atmega32u4 (which PCB companies can’t get either)

but i know that JLCPCB & co won’t order at aliexpress.
At least PCBway is willing to accept ordering parts by themselve as well as getting provided parts at the same time. But i can’t imagine how that kind of logistic stunt would work.

TeXitoi, 2021-09-27:
I think that's not a viable solution. On aliexpress, you sometime also have bad MCU. And the ones you find now are end of stock items, if you are lucky, and out of QA validation most of the time. And they are sold 5 times the official price if you are lucky. So nothing viable for a sane and affordable plan for now.