TeXitoi/keyseebee #16: why is PB12 connected to GND?

rukai, 2022-12-26:
I'm new to PCB design and have been using the keyseebee as a reference for my own board. (It will present itself as an HID keyboard and use rust)

I noticed that on the keyseebee stm32F072 pin 25 (PB12) is connected to ground while no other GPIO pins are.
Connecting to ground does allow the ERC to pass for that pin, so I can understand for that reason, but it still seems strange that only one of the pins would get that treatment.
So wanted to check if there was any deeper meaning to it other than "just is".

TeXitoi, 2022-12-26:
From my memory, that's the pin to detect the side: one is connected to ground, and on the other side, it's connected to 3.3V.

TeXitoi, 2022-12-26:
That's it: https://github.com/TeXitoi/keyseebee/blob/master/firmware/src/main.rs#L94-L95

rukai, 2022-12-26:
Ooooh thats a clever solution to that problem, thanks for taking the time to answer my question.

closed 2022-12-26 by rukai